DualTachyon/rt-4d-fw-beta #1: RT-4D blocked after beta load

Failte56, 2025-04-06:
Hi Caboom ( RT-4D  with DMR 1.2.0.6 and 3.14 FW) I've installed the Beta with Windows Radtel Flash Utility, but after installation the radio remain on "initialising DMR", to solve I've reinstalled both Radio FW and DMR FW.
Am I forgetting some steps?
  Giuseppe

DualTachyon, 2025-04-06:
Try a different flasher, in case it makes a difference. I haven't used the one you showed on Telegram.

If that still doesn't work, then revert to flashing the original firmware and I will have a look what could go wrong in the code in an hour, as I'm currently outside.

bricky149, 2025-04-06:
@Failte56 Your issue sounds like mine from a few days ago, except it was using the official updater. I don't believe it's a bug with this beta firmware. I can confirm the same behaviour on my 'broken' radio but not on my 'working' radio, which accepts this firmware with no issues.

DualTachyon, 2025-04-06:
@Failte56 try going to this [website](https://ninjaterm.mbedded.ninja/app), it lets you see messages from the UART without installing any software.

Configure it to your COM port, and then with the 4D off, press and hold the SIDE 1 button. Then turn it on.
It should print some messages and possibly some garbage. Please paste the results here.

The output should start with "HKLFBOOT" and then some more stuff.

Failte56, 2025-04-06:
HKLFBOOT......
 
 VERSION:01020006
 RF_IC:BK_4819
 RF_RX:0, RF_TX:0
 SPK_VOL_LEV:14,14 
 MIC_VOL_LEV:15,15 
 POWERSAVE_LEV:1,UVGPIOEN:1 
hhBhBh>h>?Hh%h%hhh*d"h*
hhhhhMhMhEhEhHhHhLhLh
*d"h*hk;Th}h50 @PD$4h

DualTachyon, 2025-04-06:
> HKLFBOOT......
>
> VERSION:01020006 RF_IC:BK_4819 RF_RX:0, RF_TX:0 SPK_VOL_LEV:14,14 MIC_VOL_LEV:15,15 POWERSAVE_LEV:1,UVGPIOEN:1 [...]

That looks fine... I think your problem is something else and it is possible I have a bug. Try the firmware on the previous post.

Failte56, 2025-04-06:
Advice me about the better flash tool ( Windows ) maybe I can use rt890-flash.exe ( the same for backeup)? but I nedd info on parametrs to be used

DualTachyon, 2025-04-06:
I also added a new on-screen message. It should say "Continuing" if DMR was successful, in which case the bug is for sure not DMR.

Failte56, 2025-04-06:
PTT + Power-on correct?

DualTachyon, 2025-04-06:
For firmware upgrade, yes.

Failte56, 2025-04-06:
PS C:\Users\Giuseppe\Desktop\Radtel RT-4D\Backup> .\rt890-flash.exe -4d -p COM10 -fw rt-4d-refw-dmr-test.bin
Failed to flash radio firmware  
:-(

bricky149, 2025-04-06:
> PS C:\Users\Giuseppe\Desktop\Radtel RT-4D\Backup> .\rt890-flash.exe -4d -p COM10 -fw rt-4d-refw-dmr-test.bin Failed to flash radio firmware :-(

Usually means the port couldn't be opened. I'm working on making that particular error more helpful as I made the program never throw exceptions/panics. Separate issue/bug.

DualTachyon, 2025-04-06:
You might have to try multiple times as the UART is notoriously unreliable at times. Make sure you also closed the "Ninja" website since it will have a lock on the COM port.

Also given that your issue doesn't seem to be DMR, feel free to use the first tool you originally used. Your UART capture looks good, so it's probably not the flashing tool.

Failte56, 2025-04-06:
Now I'm trying to reuse Oegatee Radtel flash utility...  Flash completed, Initialize DMR, Led red and then green.... blocked on Continuing with green led

DualTachyon, 2025-04-06:
OK, great! The good news is that your DMR chip is not broken like Bricky's and there must be a bug in my code somewhere.
Give me a few minutes to send you a firmware that will print some messages to the COM port (you will need to use the Ninja website again).

Failte56, 2025-04-06:
Shoul I restore the correct working mode before the new test?

Failte56, 2025-04-06:
ok I'll send the SDI Backup , as there is no "boot" after the upgrade and I cannot send the Ninja website, as when I powerdown and powerup the radio will restart with Database check, Initialize, continuing

Failte56, 2025-04-06:
[spi.zip](https://github.com/user-attachments/files/19621511/spi.zip)

DualTachyon, 2025-04-06:
The boot message only shows with SIDE 1, but I don't need that.
You need to power off the 4D, then click "Open" on the Ninja site and then turn on the 4D. It will print new messages I added to the new binary. The messages will just "Line XYZ, Line ABC", etc.

DualTachyon, 2025-04-06:
There is a "problem" with your "Groups". I will investigate.

[image]

Failte56, 2025-04-06:
Line 100
Line 102
Line 104

on radio screen still "continuing",,, however as you said I'll create a blank configuration

Failte56, 2025-04-06:
You mean all the entries on RX Group List?

Failte56, 2025-04-06:
Or the Group Call entries on Contacts?

DualTachyon, 2025-04-06:
> You mean all the entries on RX Group List?

Either delete the RX group list or disable it in the channel. I confirmed it is crashing during RX GROUP configuration.
I will fix it shortly.

Failte56, 2025-04-06:
I confirm , with Blank radio the new FW it's working

DualTachyon, 2025-04-06:
@Failte56 I released Beta2 binary on the main page. Please try it.

DualTachyon, 2025-04-06:
Using your SPI flash content, I am able to boot the radio normally.

Failte56, 2025-04-06:
Is it normal to receive at power on the messagges Databese check, Initialising DMR, you FW info? After that the FW boot  normally

DualTachyon, 2025-04-06:
Yes. I added those messages to help with crashes in early boot. With official firmware, such crashes can occur and my messages help to determine where it happens. Because my firmware imitates the original, it will potentially have the same bugs. So it is good to know what is the cause.

Failte56, 2025-04-06:
Thanks a lot for your Help, I'll test the FW giving feedback

closed 2025-04-06 by DualTachyon