DualTachyon/rt-4d-fw-beta #2: VFO corruption when Offset Freq is 182MHz or above

bricky149, 2025-04-06:
## Summary

I accidentally typed the TX frequency into the Offset Freq setting. This issue is also present on the stock Radtel firmware.

## Steps to reproduce

1. Go to Menu->Channel Settings->Offset Freq
2. Type in a number larger than 181MHz, then confirm
3. Back out of the menus (optional)

## Expected results

VFO should refuse the input, either with a message or silently.

## Actual results

The VFO is reset to 438.5MHz on analog, with a 88.5Hz CTC tone, High power and Wide bandwidth.

DualTachyon, 2025-04-06:
Not near my gear currently. How does one determine the valid range for the offset? Wouldn't RX=136MHz and TX=440MHz be valid?

bricky149, 2025-04-06:
I've not seen any real-world cases where someone would have out-of-band offsets, so I assumed refusing it was the better option than expanding the range in case it requires a bigger data type and breaking compatibility with how the stock firmware stores VFOs.

DualTachyon, 2025-04-06:
I don't see any obvious problems in the "apply setting" right now. I'll be at my PC soon. I don't see any range checks. It just applies the offset right away on the TX frequency. But it might do a check later to see if the resulting frequency is invalid (by TX rules).

[image]

DualTachyon, 2025-04-06:
That was exactly it. Did it happen while in Frequency Mode?

[image]

[image]

DualTachyon, 2025-04-06:
It stores the new frequency back to flash and then it issues a VFO reload. The reload sees the bad frequency and applies the settings you saw.

Kind of crappy. I'll thing of something to give better UX.

DualTachyon, 2025-04-06:
Where it sets the offset in that first screenshot, I can just do the frequency check there and return false if invalid.

bricky149, 2025-04-06:
> That was exactly it. Did it happen while in Frequency Mode?

Yes, irrespective of whether I was in analog or digital mode.

DualTachyon, 2025-04-06:
Tell me what you think of this build.

[rt-4d-refw-beta3.zip](https://github.com/user-attachments/files/19623555/rt-4d-refw-beta3.zip)

DualTachyon, 2025-04-06:
The UX is a little annoying. If you have no offset already set, you cannot choose the negative option, so you cannot try to do -300mhz at 400mhz. One has to enter a valid positive offset first, then flip the direction to negative and only then will -300 work.

bricky149, 2025-04-06:
> Tell me what you think of this build.

Works up to 176.7MHz. 176.8 and above doesn't allow me to accept it but I can cancel, keeping the last working value.

DualTachyon, 2025-04-06:
What base frequency were you at? Bear in mind that the range of valid frequencies for DMR is much smaller than Analogue. DMR is strictly 136 and 400 bands. Analogue is every except Airband and a gap or two above 500mhz

DualTachyon, 2025-04-06:
Try a +010 offset, then flip to negative, then set it as -290 (to land above 136).

DualTachyon, 2025-04-06:
I'm using the "official" method of verifying valid frequencies. The logic is in the screenshot.
I was able to set a 420MHz offset from 435Mhz.

[image]

DualTachyon, 2025-04-06:
The RangeLimits at the bottom are ranges of frequencies that you can specify in CPS as "not allowed" "RX only" "TX and RX allowed"

bricky149, 2025-04-06:
> Try a +010 offset, then flip to negative, then set it as -290 (to land above 136).

That works, thanks. When I do that with a base freq of 145.6MHz on DMR it resets the VFO.

DualTachyon, 2025-04-06:
OK, let me have a look.

DualTachyon, 2025-04-06:
Your last finding is not a bug I introduced, it was already there. Try this:

- Enter 145MHz in VFO
- Go to Offset and enter 10Mhz
- Go to Tx Freq just below, and confirm 155MHz
- Go to Offset Dir and change to -
- Go to Tx Freq and now it's 435MHz

I will deal with this tomorrow as it's late now and I'm happy it's not a bug from me.

DualTachyon, 2025-04-06:
I meant, try it with the official firmware.

DualTachyon, 2025-04-06:
I figured it out. BY changing to negative 10, the resulting frequency becomes illegal for DMR and the VFO will reset it.
The code for changing the direction also needs checks. The entire UX for Offset is just bad. The +/- should go together with the number.

DualTachyon, 2025-04-06:
I pushed a beta3, let me know how it goes for you.

bricky149, 2025-04-06:
The issue is resolved in your latest build. Thanks again.

closed 2025-04-06 by bricky149