DualTachyon/rt-4d-fw-beta #3: No RSSI on analog

Failte56, 2025-04-07:
RSSI is showing "0" when a transmission is received, I'll send a video

DualTachyon, 2025-04-07:
I will investigate. RSSI code is a bit nasty, so it might take a while to find the problem.

DualTachyon, 2025-04-07:
Can you upload the video? Here I get non 0 RSSI on any RX. The RSSI value may not be correct but it is never 0 in my case.

Failte56, 2025-04-07:
Look likr impossible to send video

DualTachyon, 2025-04-07:
Ah, that one. I was looking at the top left side RSSI. I will check the one you pointed out, I forgot it existed.

DualTachyon, 2025-04-07:
I pushed Beta 4 to the repo, can you give it a try?

There's now 2 binaries. The original Radtel UI for DMR (which will show 00000000 as ID for promiscuous, and may show sometimes broken ID with not promiscuous), and the new DMR UI for promiscuous (which has a couple of bugs when using TX/etc). You can test whichever you want.

Failte56, 2025-04-07:
Yes... signal meter is working on both releases

DualTachyon, 2025-04-07:
Great, thanks for reporting the last 2 bugs.

closed 2025-04-07 by DualTachyon