DualTachyon/rt-4d-fw-beta #4: Radio freeze when you work on " Zone Set"

Failte56, 2025-04-07:
When you use the "Zone Set" menu the radio freeze on menu selection, after some selection

DualTachyon, 2025-04-07:
I'm not sure how to reproduce this bug, I was able to scroll through zone list without issue.
But I did find a bug if i select a zone and I scroll down, I get unrelated menu options. I will fix that first.

DualTachyon, 2025-04-07:
Can you try again with the attached firmware? I don't know if the Zone problem went away, but I fixed the problem I mentioned in my reply and I hope it fixes your original issue.

[rt-4d-refw-menufix.zip](https://github.com/user-attachments/files/19629791/rt-4d-refw-menufix.zip)

Failte56, 2025-04-07:
OK, I'm just loading now

Failte56, 2025-04-07:
Now is working, just 2 selection on zone set menu, 
1 - Edit name
2 - Select CH
But on select CH there are Channel number like CH-0001, CH-002... and so on. better to have alias
I think the previous problem was due to some scrap ... as on Zone set menu there was selection like clock, sms, etc....

DualTachyon, 2025-04-07:
The CH-xxxx is official behaviour from Radtel, so I will add it to improvement TODO list.
Other than this, the Zone problem is resolved for you?

Failte56, 2025-04-07:
Yes, zone solved I think the previous problem was due to some scrap ... as on Zone set menu there was selection like clock, sms, etc....

closed 2025-04-07 by DualTachyon