DualTachyon/rt-4d-fw-beta #6: Beta Version 5 Spectrum issues

PieeerB, 2025-04-08:
The noise canceller in the spectrum opens at each signal. for 1 video spectrum on the official firmware. on the 2nd video spectrum on the beta 5 firmware.
official firmware 3.14

https://github.com/user-attachments/assets/5ecb4e44-a296-482c-89a7-adfe5b41cd04

beta 5 

https://github.com/user-attachments/assets/a0cca800-dbed-4dc8-bb6e-057263669cee

DualTachyon, 2025-04-08:
The noise + RSSI logic is a little weird, so I'm not sure if it is correct. Let me know if the attached binary seems better?

[rt4d-refw-spectrum.zip](https://github.com/user-attachments/files/19646516/rt4d-refw-spectrum.zip)

bricky149, 2025-04-08:
I've tested the binary above and it behaves like stock, which I assume is the desired behaviour.

PieeerB, 2025-04-08:
it's working well now. I found another 1 graphics pack, I'll try to show it on video.

DualTachyon, 2025-04-08:
Thanks, I will investigate shortly.

DualTachyon, 2025-04-08:
I found the bug and it is easy to fix. However I'm currently working on an Airband fix (it doesn't work properly), so it will be some time before I push an update with this visual fix. But it should be tonight.

Thanks again.

DualTachyon, 2025-04-08:
Let me know if it went away with Beta 8?

PieeerB, 2025-04-08:
> Let me know if it went away with Beta 8?

it works flawlessly, thank you

closed 2025-04-08 by DualTachyon