DualTachyon/rt-4d-fw-beta #8: Air band works worse

imedvedko, 2025-04-08:
I can't hear 129.5 frequency in AM. I can hear some sounds, but it's not clear. And I can clearly hear a pop when transmission ends. Switching between AM/SSB/FM only changes label at the bottom of the screen, it does not affect the sound

V 3.14 250321

https://github.com/user-attachments/assets/dd40711b-9bb3-4330-ba6e-9458bf0885df

beta 6

https://github.com/user-attachments/assets/ef59c77f-a0c5-45f4-a88e-dd449f07bae1

DualTachyon, 2025-04-08:
I fixed some Airband related bugs, but I'm still not getting correct audio. I uploaded Beta 8 with these fixes, let me know if anything changes for you (in my environment I only hear screeching noise).

DualTachyon, 2025-04-08:
OK I figured it out, it will take some time to fix the cause of the problem, but I know why Airband is not working now.

DualTachyon, 2025-04-08:
I uploaded Beta 9, please give it a try. It's too late currently for ATC traffic, but the bug I fixed is legitimate (but not sure if it makes AM work OK).

imedvedko, 2025-04-09:
it doesn't work :-(

beta 10

https://github.com/user-attachments/assets/09c7f317-f60d-4cdb-b822-8cf9f65b261a

DualTachyon, 2025-04-09:
Can you try the original UI binary? I think I forgot to apply the fix to the promisc one.

imedvedko, 2025-04-09:
It works as expected with new promisc beta 10 file. Thank you ;-)

DualTachyon, 2025-04-09:
Ah, great to know! Thanks for your patience.

closed 2025-04-09 by DualTachyon