DualTachyon/rt-4d-fw-beta #9: Bad displaying of "Encryption set" menu (new)

Tov17, 2025-04-08:
I found a bug with bad displaying again. Now it finds only at rt-4d-refw-beta6-promisc-ui version, but beta6-original-ui version is fine.
("Menu" - "Digital set" - "Encryption set")

Photo of promisc and original UI:
[image]
[image]
https://github.com/user-attachments/assets/7afdf425-09c3-4ee9-8990-7a3ce4b3a0e6

closed 2025-04-08 by Tov17

DualTachyon, 2025-04-08:
Can you try beta 7? I suspect I forgot to apply the fix to the beta 6 promisc...

Tov17, 2025-04-08:
Yes, i got b7 update after that i closed issue.