DualTachyon/rt-4d-fw-beta #15: Backlight error from the original firmware

PieeerB, 2025-04-11:
if the backlight is turned off, it does not turn on when an incoming signal is received, not when searching, not in VFO mod
[https://youtu.be/TB-0CaK7A3U](url)

DualTachyon, 2025-04-11:
1) Do you mean the bug is in the original firmware too, or that my firmware is different from the original?
2) Do you mean when the backlight timeout expires it goes off and then not back on, or do you mean backlight is turned off in settings? I tried briefly here and after the backlight goes off in VFO mode, any signal will turn it back on.

PieeerB, 2025-04-12:
yes, it was an error in the original firmware. yes, I set the backlight time, the time is running out, the backlight goes out and does not respond to the signal.

DualTachyon, 2025-04-12:
I'm not able to reproduce this behaviour. Can you send me a backup of your SPI flash? I wonder if it only happens under some combination of settings. If your CPS has confidential information, you just need to extract the first 32KB of SPI flash (abort the SPI backup process when reaching 2%)

PieeerB, 2025-04-12:
https://dropmefiles.com/LChXS

DualTachyon, 2025-04-12:
Thanks. I will check it in the afternoon.

DualTachyon, 2025-04-12:
You have "Save Mode" set to 1:3 and according to Radtel firmware logic, this is a feature and not a bug. I pasted a screenshot with the logic. If you want this fixed, I can add a setting later. Let me know and I will move this bug to the Discussions area for consideration later.

[image]

PieeerB, 2025-04-13:
if this is the logic of the firmware, then let it stay that way. I just turn off the energy conservation mode. It's probably worth adding an indication that the energy conservation mode is on. thanks