DualTachyon/rt-4d-fw-beta #17: Typo: change megasiemens to miliseconds.

crazy4volvo, 2025-04-11:
Could you please change all occurences of "MS" to "ms"? It hurts my eyes...

DualTachyon, 2025-04-11:
Yeah. I've been doing a cleanup of my own with some things. Less typos vs the original, and some better wording. I also want to line up the MS numbers. I don't like seeing 90 ms and 100 ms not line up at the ms.

DualTachyon, 2025-04-11:
I forgot to update here, but check latest builds, that should be fixed. I also changed "5 S" to "5 s" and so on.

Softace42, 2025-04-12:
> Я забыл обновить здесь, но проверьте последние сборки, это должно быть исправлено. Я также изменил «5 S» на «5 s» и так далее.

В версии beta17 все отображается корректно!

closed 2025-04-12 by DualTachyon

crazy4volvo, 2025-04-12:
In beta17-promisc-ui DTMF Interval and DTMF Duration are still not aligned to ms.

DualTachyon, 2025-04-12:
I queued the change to go along with next beta once I have some new bug fixed.