DualTachyon/rt-4d-fw-beta #30: Display error in dmr mode

PieeerB, 2025-04-14:
if the cursor is in vfo and vfo is being received, then the information about the dmr does not appear completely. as soon as I move the cursor to vfo b, the information is fully visible. I tried to show it on the video.

https://github.com/user-attachments/assets/b02f5f4c-03f7-4b8b-8a3e-bf7618bcab9a

https://github.com/user-attachments/assets/096b6fe3-9c42-4084-af99-dca96e40dbee

PieeerB, 2025-04-14:
yes that's right

DualTachyon, 2025-04-14:
I was able to reproduce your bug here, so now I have something I can test and investigate.

DualTachyon, 2025-04-15:
I uploaded Beta 23. Let me know if it is fixed for you.

PieeerB, 2025-04-17:
there were short radio calls in which there was no display bug. well, there were also communication artifacts, I do not know if it is related to the firmware or the local repeater. There is very little radio sharing so I can't get accurate feedback. Well, all the short radio calls that I listened to were without a display error.

DualTachyon, 2025-04-17:
I don't know what you mean with "comms artifacts", but I just found and fixed a bug where the Address Book was not used to lookup the callsign + name when using Promiscuous mode. I will push a new Beta today with this fix.

DualTachyon, 2025-04-17:
Have you had any luck reproducing the "artifacts" ? If not, does the DMR screen seem to work overall?

DualTachyon, 2025-04-19:
I'm closing this for now. You can reopen if you find the issue again. I have fixed other bugs in the meantime, maybe some of them helps remove the artifacts you saw before.

closed 2025-04-19 by DualTachyon