DualTachyon/rt-4d-fw-beta #35: Private callback to TG instead of group callback

crazy4volvo, 2025-04-15:
I noticed the issue when trying to respond to CQ on TG. Radio displays "private callback" and TG number, the hotspot dashboard says the same. Nobody hears me. Tested from beta 19 up (prom/extra).  After flashing 3.14 problem disappeared. Radio responds with Group callback and communication with TG is possible. Either bug, or something I don't fully understand.

DualTachyon, 2025-04-15:
Thanks for the report, I will investigate.

DualTachyon, 2025-04-15:
I figured it out, and it's a little headache to correct properly. There are many places where the call type is All, Private, Group. But not all places have the same value. One place A/P/G is 0/1/2, another 1/2/4, and yet another is 0/1/9. I have to double check all uses and ensure they're the correct value. I'll fix it tomorrow.

DualTachyon, 2025-04-15:
It seems I only have the values wrong in the "Call hold" (callback) code. While fixing it, I think Radtel themselves have a bug in how to configure the call. While it doesn't have any noticeable side effects, it is effectively sending the wrong data to the DMR, who happens to interpret it the correct way anyway.

DualTachyon, 2025-04-15:
Can you try this binary?

[rt4d-refw-callback.zip](https://github.com/user-attachments/files/19766759/rt4d-refw-callback.zip)

crazy4volvo, 2025-04-16:
Many thanks, I'll give feedback in the evening (now going to QRL).

crazy4volvo, 2025-04-16:
OK at least with a repeater, hotspot tests in a hour or two.

DualTachyon, 2025-04-16:
Great! I will include this change to the Beta I prepared for tonight. Closing the bug, re-open if you find it's not fully fixed.

closed 2025-04-16 by DualTachyon