DualTachyon/rt-4d-fw-beta #38: Impossible to put Area B in Zone Mode - beta25-extra

crazy4volvo, 2025-04-17:
It's impossible to put Area B in Zone Mode neither from menu Basic->Area B Show, nor with hotkey "#". Area A behaves normally.

DualTachyon, 2025-04-17:
Can you make a video? I'm unable to reproduce this problem here.

https://github.com/user-attachments/assets/cc6382d8-c55a-4a90-9e49-6eb267297367

David5466, 2025-04-17:
This happened to me too, I think it did not like my long zone name, and refused to show the zone in " B " 
So I shortened the Zone Name from 477MHz CB RADIO to UHF CB RADIO, then it liked that and everything worked good

DualTachyon, 2025-04-17:
Zones are annoying because they have a concept of "current channel" from the 200 it can hold, so there can be inconsistencies if you now display the same zone on VFO A and B in Dual Display mode. You change the channel in B, sometimes it shows in A sometimes it doesn't.

It is an area I need to investigate because I don't like inconsistencies. Thinking loud, I feel like the same zone shouldn't be used twice, but I haven't thought about it too deep

DualTachyon, 2025-04-30:
Can this one be closed?

DualTachyon, 2025-05-01:
> Can this one be closed?

Ping :)

closed 2025-05-02 by DualTachyon