DualTachyon/rt-4d-fw-beta #39: Basic->Zone Set causes havoc in Zone definitions

crazy4volvo, 2025-04-17:
When trying to edit Zone Names and Channels varoius things happen:

- once assingned channels sometimes cannot be unassigned
- Zone Names disappear and cannot be set
- CPS shows multiple instances of the same Channel assigned to Zone in random order

Tested with beta 24,25 and 26 extra.

DualTachyon, 2025-04-17:
Can you also test on the official firmware? It will help narrow down where I need to look for if the bug is mine and not theirs. (I'm currently out for the next hour)

crazy4volvo, 2025-04-17:
Sure, but in QRL I'm on linux and my previous attempt to flash original .bin failed. So I'll test it probably  in the evening.

crazy4volvo, 2025-04-17:
OK, no risk - no fun. V3.14 has errors too. It was able to edit and save Zone Name, but assigning Channels is buggy. Do we really have only 1024 channels? ;P

[image]

crazy4volvo, 2025-04-17:
Confirmed - buggy Zone editing in 3.14. I started with clean codeplug  and defined 1 Zone and 6 Channels. After some play with adding and removing Channels to Zone (sometimes it was impossible to remove Channel) it ended up with Channels 2,5,4,6,4 added to Zone. Duplicating Channels is impossible in CPS.

DualTachyon, 2025-04-17:
OK, I'll investigate.

DualTachyon, 2025-04-19:
I didn't have time to look at this today, but can you try my latest beta to see if there any area of improvement with zones? I fixed 2 bugs, one mine, one Radtel.

DualTachyon, 2025-04-20:
@crazy4volvo can you retest your zone issues with this build? https://github.com/DualTachyon/rt-4d-fw-beta/discussions/10#discussioncomment-12890717

I fixed some bugs I found while doing other stuff. I'm preparing a Beta 29 for tomorrow, which will include some zone fixes.

crazy4volvo, 2025-04-21:
Hi, I can't do it right now as i'm on my holiday trip to Germany and unfortunately forgot the programming cable...

crazy4volvo, 2025-04-28:
In beta31 adding and deleting channels to/from zone seems OK.  There is another issue - when entering into zone editing (name, channels) the first zone is de facto edited despite displaying current zone.

Steps to replicate an issue: 

- set area to Zone Mode
- assign 2nd, 3rd etc Zone to Area
- enter Menu / Zone Set (the correct 2nd, 3rd etc Zone Name is selected with arrow).
- enter this Zone
- enter Edit Name (or step back with red key)  -> name of 1st Zone is edited/displayed. 

We have to explicitly choose the Zone with cursor keys for editing.

DualTachyon, 2025-04-28:
I was able to reproduce it, I'm looking into it.

DualTachyon, 2025-04-28:
This looks like a Radtel bug, fun... My 4D with official 3.14 has the same problem.

DualTachyon, 2025-04-28:
I fixed it, but I need to double check I'm not introducing another problem as the fix is not an usual pattern in this firmware.

https://github.com/user-attachments/assets/d730fe42-3daa-4437-91c6-a60e6e7bb774

DualTachyon, 2025-04-28:
Give this a try?

Please note this build is based on the 3.15 baseline and it means your 3.14 RX groups will be corrupted except for the first one. Radtel recommends starting with a fresh CPS and reloading channels CSVs etc and redoing settings. Some features of 3.15 were not imported because I have some doubts of their new DMR logic and their spectrum also offers a worse UX. The majority of the changes are in.

[rt4d-refw-2025-04-28.zip](https://github.com/user-attachments/files/19943378/rt4d-refw-2025-04-28.zip)

crazy4volvo, 2025-04-30:
2025-04-28 edits zones correctly, thanks.

DualTachyon, 2025-04-30:
> 2025-04-28 edits zones correctly, thanks.

Great. Are all your concerns addressed with the .bin ?

crazy4volvo, 2025-04-30:
Hmm, it seems zone names are edited correctly, but old bug from the first post (duplicates, unable to unassign) is back :(.  Version rt4d-refw-2025-04-30-dtmf4.

DualTachyon, 2025-04-30:
> Hmm, it seems zone names are edited correctly, but old bug from the first post (duplicates, unable to unassign) is back :(. Version rt4d-refw-2025-04-30-dtmf4.

Can you find a beta build (from Issues or from release page) where that was working? I've lost track of what I fixed before with the (massive) 3.15 upgrade.

DualTachyon, 2025-04-30:
> Can you find a beta build (from Issues or from release page) where that was working? I've lost track of what I fixed before with the (massive) 3.15 upgrade.

And ideally a video (or a link to a post with one) of the problem.

crazy4volvo, 2025-04-30:
Ten posts above - in beta31 adding and deleting channels was fixed. It was only wrong (first) zone selected for editing unless it was explicitly selected.

DualTachyon, 2025-04-30:
> Ten posts above - in beta31 adding and deleting channels was fixed. It was only wrong (first) zone selected for editing unless it was explicitly selected.

I git diff'ed beta31 and what I have now, there are no differences related to zone code. I think you have some different problem. Can you try reproducing the bug from a clean/fresh CPS, manually adding a few random channels and editing a zone? I would love to see a short video clip of the issue happening.

DualTachyon, 2025-05-01:
For historical records and based on the tests and binary in Telegram, do you think this issue is now resolved?

crazy4volvo, 2025-05-01:
Yes,  seems solved in 2025-05-01-4.bin. Many, many thanks.

DualTachyon, 2025-05-01:
Thank you for finding the bugs and the nice repro video!

closed 2025-05-01 by DualTachyon