DualTachyon/rt-4d-fw-beta #50: Problem with DMR transmission in Dual Standby mode

DSSlan, 2025-04-25:
Hello, 
@DualTachyon, I'm writing here because I noticed a long time ago that the RT-4D has a very serious bug that Radtel's programmers couldn't handle (I wrote to their support many times). 
It's about a bug that is constantly there, also when using your firmware, because, as I suspect, it's in the DMR firmware part. This bug manifests itself in the fact that if the Dual Standby (Dual Watch) function is enabled, sooner or later the transmission in DMR "suspensions" occur. 
To be precise: the radio doesn't stop transmitting completely in DMR, but the repeater (HotSpot too) doesn't react to this transmission - it looks as if some data (e.g. DMR ID or Color Code) was missing in the signal during transmission. 
This "suspension" occurs at completely random intervals: after a few minutes, hours or right after the radio is turned on. 
To fix this, you can do two things: either turn off the radio's power for a moment, or wait until someone starts transmitting on a given DMR channel - then the radio sort of "unblocks" and transmits again (until the next "hang" after a random timeout).
Interestingly, I haven't found a single piece of information about this problem anywhere on the Internet, and it's not an isolated case, because I checked it on three different RT-4Ds (in every DMR firmware version, including the latest 1.2.0.6).
Have you noticed this bug? Did you pay attention to it?

DualTachyon, 2025-04-25:
I cannot TX, so it's not something I would have noticed. I use the 4D as a scanner/RX unitm. Testers on Telegram do the TX'ing for me.

Someone said they might have had the issue you report but they weren't in a position to debug/investigate at the time.

You're probably correct that it is likely to be the DMR firmware. I can check the DMR documentation later today, to see if there is any kind of soft reset or other clean up operation that could help troubleshoot this issue.

I will check the DMR command séquence involved in dual standby to see if anything catches my eye.

DSSlan, 2025-04-25:
Thank you very much for your quick response.

The RT-4D is a radio with undoubtedly great potential, and it would be very nice if you could find the cause of this annoying problem.

At the same time, thank you for how much you have already done to remove factory bugs in the firmware and add new options. You are great :)

I am eagerly awaiting for some information regarding the problematic Dual Standby.

Best regards

DualTachyon, 2025-04-25:
One thing you can try is turn off Save Mode, if only to get rid of weird power saving mechanisms that kick in regularly.
It might not fix a thing, but it's a couple of commands less to worry about.

DualTachyon, 2025-04-25:
The dual standby sequence is fairly simple: Spend some X time in analogue, some Y time in DMR and repeat. This assumes no save mode (which adds extra timings).

Going to analogue puts the DMR into deep sleep mode and redirects the BK4819 to the MCU.
Going to DMR redirects the BK4819 to the DMR chip, wakes up the DMR, sets up the typical configuration (DMR ID, Channel information, group list, key), optionally enables promiscuous mode and turns off DMR power saving mode. I don't see anything special in this aera.

There could be 3 things:

1) the 4D is doing a real transmission but the parameters are wrong - I asked for help for someone to try and capture DMR traffic if they encounter the problem
2) the 4D wants to transmit, but it is struggling to configure the BK4819 (in DMR mode, BK4819 is fully controlled by the DMR chip). The switching logic is fairly simple, I don't see anything that could be wrong. However maybe some extra delays could be needed (currently there's a 50ms delay DMR->ANA, and 10ms delay ANA->DMR).
3) the 4D locked up / has some weird bug, or needs a soft reset. This is the trickier one to handle.

The DMR doc is in Chinese, so it will take some time to find anything that might help with this.

DSSlan, 2025-04-25:
Okay. Thank you very much for the information and tips. Just in case, I will indeed turn off the power saving mode for now (I have been using 1:1 so far) and observe...

DSSlan, 2025-04-25:
Unfortunately, turning off Safe Mode did not help. The problem with transmitting appeared after a few minutes. It was necessary to turn off the power for a while :(

DSSlan, 2025-04-25:
How long do I wait before TX? These are irregular intervals: every few minutes, every half hour, every hour; it varies.

I have ChA set to analog mode, and ChB to DMR. Of course, for testing, I can swap them around to see if it makes any difference.

DualTachyon, 2025-04-25:
In terms of code, A and B are the same.

DualTachyon, 2025-04-25:
Can you try with 2 DMR channels? The switching logic is lighter when the RF mode remains the same between A and B. It's worth trying with a dummy frequency so that it never really finds any traffic just to see the problem exhibits itself.

DualTachyon, 2025-04-25:
Radtel also has a new DMR firmware coming out soon. It's 1.2.0.12 and here's part of what they told me during a technical conversation:

"The update from 1.2.0.8 to 1.2.0.12, as far as I know, only **fixes the problem of failing to connect to the repeater**, and all-call SMS is enabled again."

DSSlan, 2025-04-25:
Hmmm... but how should we understand "failing to connect to the repeater"? Isn't this about solving "our" problem? After all, this is the problem that occurs when connecting to a DMR repeater/HotSpot. Unless I'm interpreting it wrong.

DSSlan, 2025-04-25:
> Can you try with 2 DMR channels? The switching logic is lighter when the RF mode remains the same between A and B. It's worth trying with a dummy frequency so that it never really finds any traffic just to see if the problem exhibits itself.

Of course I can do that. I'll set both channels to DMR mode. I'll also set frequencies where no one will transmit. I'm setting that right now and I'll let you know how the radio behaves then.

DSSlan, 2025-04-26:
Update after about 11 hours:
with Safe Mode turned off and DMR mode in both: ChA and ChB, transmitting works fine for now. I am still testing, however, because previously it happened that the "freeze" would appear even after a dozen or so hours.

DualTachyon, 2025-04-26:
Grab the latest CPS from Radtel, it includes a new firmware 3.15 and new DMR firmware 1.2.0.12.
Where you want to use my beta or the 3.15 main firmware is up to you. 1.2.0.12 is compatible with my firmware.

DSSlan, 2025-04-26:
OK. I have just downloaded and installed DMR version 1.2.0.12.
Of course I will stick with your beta 31 firmware, because I think it is simply better than the one from Radtel.

So I am testing Dual Standby on the new DMR firmware. Well, we will see how it goes now :)

DualTachyon, 2025-04-26:
You should try the new 3.15 beta also, they have updated the DMR handler code.

In the meantime, I will be busy upgrading my code base against the 3.15 baseline to inherit any improvements/fixes they did to DMR, in case it makes a difference.

DSSlan, 2025-04-26:
I think you are right and I also thought the same thing. I will be installing Radtel v3.15 right now.

DualTachyon, 2025-04-26:
I just saw additional handling for report status and promiscuous, so please try changing promiscuous mode (if it was on, try it off, and vice versa).

DualTachyon, 2025-04-26:
There are also timing changes...

This is 3.14
[image]

This is 3.15

[image]

DualTachyon, 2025-04-26:
Ghidra is showing me many small differences here and there, so there's bound to be more changes I don't know about.

DSSlan, 2025-04-26:
I feel sorry for you, @DualTachyon, because it looks like you have a lot of work ahead of you again. 

v3.15 from Radtel installed.

I had Promiscuos mode set to "Off" on each channel - I changed it to "On" as you suggested. 

By the way: I had to reconfigure the "RX Group List" because almost all entries disappeared after installing DMR v1.2.0.12 and using CPS version 1.22 :)

DualTachyon, 2025-04-26:
There is a readme in Chinese in the archive, look in there for changes. They say you have to redo the codeplug

DSSlan, 2025-04-26:
Unfortunately, I immediately noticed a very serious bug in version 3.15 from Radtel.

I didn't even have time to check whether transmitting in DMR would work correctly here, because in this version, receiving works terribly. 
The problem is that while listening, it happens that despite the fact that the correspondent is transmitting, the green diode lights up, but the correspondent cannot be heard at all; there is also no information on the display - as if no one was transmitting.
When the correspondent stops transmitting and then starts again, everything works correctly, i.e. you can hear him and the appropriate information appears on the display.
A moment later, it is wrong again and again only the green diode lights up, but nothing else happens. 

So it looks like Radtel's programmers, by introducing corrections to the firmware, broke something along the way. 
It works so badly that version 3.15 is completely unusable, because you can't hear every second correspondence! 

I'm reinstalling beta version 31 from you, because v3.15 from Radtel is a disaster :(

DualTachyon, 2025-04-26:
That is the "DMR activity" indicator. When a DMR channel has something going on (DMR, Tier3 control, etc) it will light up the green LED. It is normal.

DSSlan, 2025-04-26:
Yes, yes, I know, but that's not what I mean. 
Maybe you misunderstood me. 

For example:
I observe on the HotSpot Dashboard that two colleagues are talking and I see them transmitting on the Dashboard. At the same time, I listen to them on the radio.
And now:
- colleague one is transmitting, I see it on the Dashboard and hear it on the radio - everything is OK;
- colleague two is transmitting, I see it on the Dashboard and hear it on the radio - everything is still OK;
- colleague one is transmitting again and... Attention! You can see it on the Dashboard, but you can't hear it on the radio at all, nor can you see the relevant information on the display; only the green diode is lit;
- colleague two is transmitting again and here everything is OK;
- colleague one is transmitting again and again everything is OK;
- colleague two is transmitting and it's OK;
- colleague one is transmitting and again you can't hear him, although the transmission is visible on the Dashboard and the green diode is lit on the radio.

I mean the above behavior. The radio simply does not respond properly to every second or third correspondence - differently, sometimes more often, sometimes less often.

Have I described it clearly now?

DualTachyon, 2025-04-26:
Hmm, strange.

DSSlan, 2025-04-26:
Well. I went back to your beta 31 version, but I left DMR v1.2.0.12 installed.
As I wrote above: there really is something wrong with version v3.15. I had constant problems with every second or third transmission not being received and the radio would freeze up over and over again, which required a power restart. 
Of course, it is not impossible that the above problems are due to some "garbage" that remains somewhere in the memory, but the radio (or CPS) does not have a function like "Reset to Factory default", so what else could I do? :)

DualTachyon, 2025-04-26:
Radtel recommends writing a blank CPS to the radio and then restore contacts/channels.
There is definitely a conflict with their new firmware vs old.

DualTachyon, 2025-04-26:
Basically, launch CPS then write right away without loading or setting anything.

DSSlan, 2025-04-26:
Okay. I reinstalled v3.15 and DMR v1.2.0.12, and then sent an "empty" CodePlug to the radio using CPS 1.22.
I understand that "and then restore contacts/channels" means that after sending an "empty" CodePlug to the radio, I should simply load the previously saved configuration from the .4rdmf file to the CPS and send it to the radio, because that is what I did. 

Unfortunately, the radio behaves exactly the same as before: after a few transmissions (2-3), the radio stops receiving (nothing can be heard on the display, no information about the transmission) and only the green diode lights up. 

Interestingly: if I switch "Promiscuos mode" from "Off" to "On" and vice versa, the radio starts receiving correctly, but again only for the next two or three transmissions, and then stops receiving again (the green diode lights up). 

I will try to do one more thing: I will reload the "empty" CodePlug into the radio again and manually configure only one channel for testing (I will not restore anything from the saved backup file of channels/contacts) - such an absolutely clean, basic configuration and we will see what happens then.

DualTachyon, 2025-04-26:
The old .4rdmf will contain "bad" settings. I think you are meant to restore from CSV.

DSSlan, 2025-04-26:
I did as you wrote. I sent the "clean" CodePlug to the radio, but with imported contacts and channels from CSV files.
I also think that the backup in the .4rdmf file may contain some "garbage".

OK. Let's test... :)

DSSlan, 2025-04-27:
After several more hours of testing in DMR, I think I can confirm that the cause of both reception problems (no audibility of every second or third correspondence, up to complete lack of reception) and transmission problems (no HotSpot/repeater response to the radio carrier) is the enabled Dual Standby mode. 
With Dual Standby disabled, the radio behaves correctly, i.e. I did not notice any problems with both reception and transmission. 
Well, maybe with one small exception. I noticed that despite setting the operation on a given channel to Time Slot 1, the radio can sometimes transmit on Time Slot 2 and it does not matter here whether I set the "DMR Mode" on a given channel to "Dual-slot off" or to "Direct Dual-slot". I think that in versions lower than 3.15 there was no such behavior of the radio, but I could have missed it.

Returning to the negative impact of the enabled Dual Standby function on reception and transmission in DMR: what next, @DualTachyon? Do you have any idea what other tests I could do? I guess I have to wait until you find some reason/error in the software code :)

DualTachyon, 2025-04-27:
I need more time to finish the 3.14 <> 3.15 diffing, I'm almost done. Then I want to analyse properly what they have done, there are some non negligible changes to the DMR state machine. Currently I don't have any new ideas other than making sure the exhaustion of different settings to be sure it is not impacted by something seemingly unrelated.

DualTachyon, 2025-04-27:
I have also heard from a few people that 3.15 seems unreliable when it comes to using different TS and groups.

DualTachyon, 2025-04-28:
I don't expect any improvements for your case, but I would appreciate if you could test my current build based on 3.15: https://github.com/DualTachyon/rt-4d-fw-beta/issues/39#issuecomment-2835779042

Check the comment I left there since RX groups are affected.

DualTachyon, 2025-04-28:
I have one last patch with just the new DMR logic, I will be working on it next.
FYI, someone brought up your same issue with Radtel CS on Telegram, and they said they were trying to reproduce it.

DSSlan, 2025-04-29:
OK. I just installed your beta 32 and the first thing I noticed was an automatic change in the settings of each DMR channel. I mean, in each channel I had the "ID Select" option set to "Radio ID", meanwhile when I loaded your firmware, "ID Select" changed to "Channel ID" and I had to change all several dozen channels back to "Radio ID". Well, nothing - luckily I didn't have that 1000 :D
OK. So I'm testing beta 32...

DualTachyon, 2025-04-29:
Hmm, that is so strange because I didn't change anything with TS handling on TX.

I attached a new binary that now includes the 3.15 DMR logic. It changes how calls are triggered. It mostly affects how RX is handled, but there could be some butterfly effect to TX, I don't know. They also introduce potential for race conditions that I think will trigger other bugs (e.g. I change from channel 100 to 101, and a second later I get an RX call from 100... ???). I had people report weird bugs on Telegram. Many were user environment issues rather than firmware but a fair few weren't. I fixed a couple of bugs with the display of CC, and a Radtel bug where non-promiscuous will open RX when a different CC comes in!

The jury is still out there on 3.15 in my opinion. I'm also hoping Radtel will have reproduced your issue and a couple others and give us some feedback.

For now I don't have a solution for your issue. I want to try adding some form of locks/mutexes around the DMR state machine that I suspect might be the cause of some of today's weird bugs. Maybe it will help you, maybe it won't, but having race conditions certainly won't help. It will take a day or two to make this happen.

[rt4d-refw-2025-04-30.zip](https://github.com/user-attachments/files/19968802/rt4d-refw-2025-04-30.zip)

DSSlan, 2025-04-30:
_"I fixed a couple of bugs with the display of CC, and a Radtel bug where non-promiscuous will open RX when a different CC comes in!"_

I confirm the above: I also had a similar bug on my radio :) 

Of course, I will install your new firmware soon and continue testing. Don't worry at all that this strange thing happens to my TX when "Dual Standby" is on. If this bug keeps recurring, I'll use the radio with "Dual Standby" off and will waiting patiently for any solution. 
Kudos and thanks to you for correcting so many bugs :)

DSSlan, 2025-04-30:
After newest beta 32 installed: 
- TX in DMR with "Dual Standby" is still not working properly ("freeze" after random time), but here we are waiting properly :) 

I'm afraid, I found one another bug: 
- I have set the channel to TG91 (World) on TS1, but I hear TG260 (Poland), which I have set on this TS. Promiscuos Mode is Off.

DualTachyon, 2025-04-30:
Can you confirm your RX group does not include TG260? I have been testing the filtering yesterday with success.

DSSlan, 2025-04-30:
Yes, I confirm: RX group include only TG91, not include TG260.

DualTachyon, 2025-04-30:
Can you try with dual standby off? I have a feeling that the promiscuous notifications are polluting the DMR Logic.

DualTachyon, 2025-04-30:
This is driving me crazy. Just yesterday it was working fine ..

DualTachyon, 2025-04-30:
I figured it out... I somehow deleted a break statement. I'll fix it.

[image]

DualTachyon, 2025-04-30:
Try this build for the non-promiscuous mode. It should be fixed. This build has some new changes that may or may not interest you. I'm preparing to release it tonight.

1) Press the green key on the "DTMF Select" strings to send that DTMF (request to easily unlock Allstar nodes, etc)
2) Added "Send DTMF" to the hot key list to quickly do 1)
3) New hot key "Next Zone" to select the next zone
4) You can block frequencies/channels during "scanning". Side 1 key to block, long press Side 1 key to clear the block list. The block list is cleared when exiting scan mode.
5) Press the menu key in the dial screen to dial (same as pressing PTT).

[rt4d-refw-2025-04-30-dtmf3.zip](https://github.com/user-attachments/files/19981440/rt4d-refw-2025-04-30-dtmf3.zip)

DualTachyon, 2025-04-30:
6) You can also bind "Long press Green key" like the other keys, not just "Contacts" / "Address book".

DSSlan, 2025-04-30:
> Try this build for the non-promiscuous mode. It should be fixed. This build has some new changes that may or may not interest you. I'm preparing to release it tonight.
> 
>     1. Press the green key on the "DTMF Select" strings to send that DTMF (request to easily unlock Allstar nodes, etc)
> 
>     2. Added "Send DTMF" to the hot key list to quickly do 1)
> 
>     3. New hot key "Next Zone" to select the next zone
> 
>     4. You can block frequencies/channels during "scanning". Side 1 key to block, long press Side 1 key to clear the block list. The block list is cleared when exiting scan mode.
> 
>     5. Press the menu key in the dial screen to dial (same as pressing PTT).OK
> 
> 
> [rt4d-refw-2025-04-30-dtmf3.zip](https://github.com/user-attachments/files/19981440/rt4d-refw-2025-04-30-dtmf3.zip)

OK - I'm installing it...

Jumbotroll, 2025-04-30:
Hi. I am testing this version and noticed that in the menu: DMR Version 38 is blank.
Regards Jorn

DualTachyon, 2025-04-30:
> Hi. I am testing this version and noticed that in the menu: DMR Version 38 is blank. Regards Jorn

Thanks. It happens here too. I'll fix it shortly.

DualTachyon, 2025-04-30:
> > Hi. I am testing this version and noticed that in the menu: DMR Version 38 is blank. Regards Jorn
> 
> Thanks. It happens here too. I'll fix it shortly.

> Hi. I am testing this version and noticed that in the menu: DMR Version 38 is blank. Regards Jorn

Here's a fixed version.

[rt4d-refw-2025-04-30-dtmf4.zip](https://github.com/user-attachments/files/19982614/rt4d-refw-2025-04-30-dtmf4.zip)

Jumbotroll, 2025-04-30:
> 
> Here's a fixed version.
> 
> [rt4d-refw-2025-04-30-dtmf4.zip](https://github.com/user-attachments/files/19982614/rt4d-refw-2025-04-30-dtmf4.zip)

Thanks! Installed and Working.
I love the Next Zone Key function!
Great work 👍

MIF666, 2025-05-01:
One small mistake. Termination beep does not go off after a DMR session. (digital set/call tone)

DualTachyon, 2025-05-01:
I fixed it, but ideally and next time, new/unrelated bugs not from the original poster should really be in new bugs as it makes it harder for me to know which bugs are tracked where.

[rt4d-refw-2025-05-01-1.zip](https://github.com/user-attachments/files/19999327/rt4d-refw-2025-05-01-1.zip)

DualTachyon, 2025-05-05:
I want to restart this conversation now that some of the chaos I faced is gone and I don't have so much on my plate right now. I can dedicate some time to make builds for you to test.

Can you remember what type of TX you have done? Was it always for example "replying to a TG", or maybe it was "I keyed in a new TG", or it seems to happen regardless of how you TX'ed? The reason I ask is there are multiple logic checks to deal with TX and the way DMR is resumed/restarted/etc. Also, were you getting RX the moment you started to TX? Or does it happen even when the 4D is "silent" for a while and then you decide to TX?

One theory I have is that the DMR logic is missing locks (like a mutex) for certain operations. For example, I have seen the wrong callsign being displayed after some time and I'm almost sure I know why it happens. I did some workaround for that but to reproduce the problem in the first place is "hard".

Depending on how you answer my multiple questions, I may want to add some locks to clear certain info / prevent some action from happening too early, to see if it would help your case. It would also let us know whether it's really a DMR problem or a main firmware problem. While the issue might appear to be DMR firmware, it could be that the main firmware is sending commands too fast or in a bad order. While this would still be a bug in the DMR, it would be information we can give Radtel to take it up to the vendor.

DualTachyon, 2025-05-05:
> OK. I will try to answer your questions precisely.
> 
> _**"Can you remember what type of TX you have done? Was it always for example "replying to a TG", or maybe it was "I keyed in a new TG", or it seems to happen regardless of how you TX'ed? The reason I ask is there are multiple logic checks to deal with TX and the way DMR is resumed/restarted/etc.".**_
> 
> As for the "freezing" of transmission, it happened completely regardless of whether I was the one responding to someone's call or whether I was the one who called first. It doesn't matter. However, I will honestly write that there were not very many attempts to conduct conversations during which the "freezing" occurred - maybe a dozen or so. However, even during such a conversation (alternating TX and RX), the "freezing" would suddenly occur and my interlocutor would stop hearing me. Sometimes, however, the entire correspondence worked fine until the end (meaning that my TX did not "freeze"). However, since I did not want to bother my colleagues with such continuous tests, most of these tests looked like this: I would turn on the radio and it would remain in listening mode, and from time to time I would press PTT and observe on my HotSpot, two repeaters, and the BrandMeister website whether my transmission was "visible". I used various TGs, but most often three: Polish 260, Jurassic 260950 and world 91. Transmission would "suspend" at completely different intervals. For example, it could happen already a few minutes after turning on the radio, but it also happened so many times that I left the radio on all night, and when I checked in the morning after waking up, TX was working properly (after so many hours). However, after a few or a dozen more minutes, the radio would stop TX again at the next attempt. I suspect that this is important, which is why I am drawing attention to the fact that during the tests, ChA was almost always set to analog mode, and ChB - to DMR mode (or vice versa). I did some tests in which both ChA and ChB were in DMR mode and then I did not notice any problems with "hanging", so I focused primarily on Dual Standby in ChA analog and ChB DMR mode (or vice versa). What is also important, transmission in DMR never "hanged" if Dual Standby mode was disabled.
> 
> **_"Also, were you getting RX the moment you started to TX? Or does it happen even when the 4D is "silent" for a while and then you decide to TX?"_**
> 
> I do not know if I understood correctly what you mean in this question. I understand that you are asking whether if TX "hanged", RX continued to work - yes, RX worked all the time. And as I wrote above: it did not matter at all whether I transmitted immediately after receiving correspondence or after a longer period of "silence". It happened at completely random intervals. TX sooner or later "hanged" always, but RX always worked.
> 
> If any of my translations are unclear to you, feel free to ask :)

It feels more and more like the switching of the RF chip between analogue and DMR could be an issue, specially with your comment that DMR <> DMR dual standby doesn't appear to show problems.

I've made a new build of beta 33 (not yet released) that adds some extra delays in between the switches and also right after wake up of the DMR, just in case things need to settle.

To any other people seeing this new build and finding bugs, please file new bugs. I won't bite.

[rt4d-refw-2025-05-05-dsslan.zip](https://github.com/user-attachments/files/20041636/rt4d-refw-2025-05-05-dsslan.zip)

DualTachyon, 2025-05-05:
> It feels more and more like the switching of the RF chip between analogue and DMR could be an issue, specially with your comment that DMR <> DMR dual standby doesn't appear to show problems.
>
> I've [...]

If you're curious, try pressing the # a few times in the DMR dial screen (green key) while under Private, and a new hotkey feature (press green key after detecting a frequency) :)

DSSlan, 2025-05-05:
I don't know what to say. You're just awesome :D

OK, thank you very much - I'll install and test right away.

Just in case, I'll try both with Safe Mode off and on, although in previous tests (which I forgot to mention above) Safe Mode doesn't seem to have any effect on "hanging" the TX.

DualTachyon, 2025-05-06:
> I don't know what to say. You're just awesome :D
> 
> OK, thank you very much - I'll install and test right away.
> 
> Just in case, I'll try both with Safe Mode off and on, although in previous tests (which I forgot to mention above) Safe Mode doesn't seem to have any effect on "hanging" the TX.
> 
> --
> 
> Installed. Testing start...

Well guess what? Remember I said something about timings? Radtel released 3.16 today and it includes timing changes that I can see will prevent collision/race conditions! I don't know when I will finish 3.15<>3.16 comparison, but some of the changes are promising.

[49 more replies]