DualTachyon/rt-4d-fw-beta #54: Beta32: screen artifacts when exiting from alarm in analog mode.

Tov17, 2025-05-03:
When you switch down the analog alarm, the screen has wrong drawing. Screen does normalize after popup norification also after going into menu.

FW is beta-32, DMR is 01.02.00.12

[image]

DualTachyon, 2025-05-03:
Thanks, I will look into it. Could you look at some of the discussion threads where I left you some feedback? E.g. Online ID checker and a couple of others.

DualTachyon, 2025-05-03:
Try this build?

[rt4d-refw-2025-05-03-corruption.zip](https://github.com/user-attachments/files/20024550/rt4d-refw-2025-05-03-corruption.zip)

Tov17, 2025-05-04:
Test confirms that issue is solved. Thanks!

Tov17, 2025-05-04:
> Thanks, I will look into it. Could you look at some of the discussion threads where I left you some feedback? E.g. Online ID checker and a couple of others.

Thank You for reminding, i wrote some messages)

closed 2025-05-04 by Tov17