DualTachyon/rt-4d-fw-beta #56: Contact ID vs radio id

mprattmd, 2025-05-05:
Using latest CPS from Radtel, using beta 32 on radio.
I have contact id set to 1 and Radio ID set to my radioid.net user
I have the channel set to look at radio but it is sending a 1 as the ID
If I go into the radio and change the channel setting to "radio id" again, it seems to work

DualTachyon, 2025-05-05:
I think there might be some confusion here... What do you mean with "Contact ID" set to 1? If you are referring to the "Channel ID" in CPS, that is not a contact ID but a DMR ID. When you change "ID Select" in CPS, you are telling the radio whether to use the global DMR ID (= radio id) or a per-channel DMR ID (= channel ID). Basically, you set a global DMR ID in the "Radio Name and ID" and if you have a repeater/channel where you need to use a different ID (e.g. a hotspot or a different DMR network), you can set that different ID in "Channel; ID" and configure "ID Select" accordingly.

[image]

DualTachyon, 2025-05-05:
In addition, CPS often messes up the settings when writing. I recommend you write your settings, let the radio reboot, then read them back and fix any bad fields that might have changed, then write it all back again. This is a problem with CPS and not the firmware. This happens even on the official one.

mprattmd, 2025-05-05:
yes, if you select RadioID, it is supposed to pull in the one defined in "Radio" tab
I saw a comment on the Facebook group where you have to select that dropdown at least once on the channel for it to recognize it, and as you said, looks to be a cps issue. 
Nice job on the firmware, great little radio.  I have Tait and Motorola DMRs and these are amazing audio quality for such a small price point.

DualTachyon, 2025-05-05:
Radtel were made aware several times of the CPS issues, but it's going to take time. I reported several bugs in 3.14 to them, and it still took them a while to release 3.15 with only half the bugs fixed.

closed 2025-05-08 by DualTachyon