DualTachyon/rt-4d-fw-beta #58: DMR timeout

mprattmd, 2025-05-10:
3 DMR issues I am seeing on beta 33

1. DMR times out.  After about 10 minutes, if no key up has occurred, the connection just drops while listening
2. The CallerID and GroupID is not displayed correctly I have tried setting display to both of these and the group shows weird characters and the caller shows only partially. 
3. Talker Aliases are also not displayed if set to callerID

DualTachyon, 2025-05-10:
1. That might be a DMR firmware problem. I never receive any RX call that lasts that long. I would suggest you test it with the official firmware also, it will tell me whether it's a DMR firmware bug or main firmware issue.
2. I need photos or videos of what you observe. I haven't heard or seen reports of this. There is a current bug where the group ID will show a long number and an equal sign, like [here](https://github.com/DualTachyon/rt-4d-fw-beta/issues/50#issuecomment-2863020288). Most likely a DMR firmware bug.
3.  Talker Alias is not supported by the current DMR firmware and unless the DMR vendor changes its mind, it's not coming for now.

DualTachyon, 2025-05-10:
Regarding 3, if you're talking about just callsign information rather than DMR TA (a different thing), check your setting for "Caller show". This toggles what you see on the screen. Also, if some ID doesn't show a callsign, usually that means your address book is out of date and you need to grab a new one.

mprattmd, 2025-05-10:
I don't mean a continuous audio I mean ime en if listening to a back and forth convo

 attached the screen of what I see on the id screen

Mark Pratt, MD
615-336-5473

DualTachyon, 2025-05-10:
For your first problem, I would test it with official firmware first. Then check if your repeater or hotspot doesn't require a regular key up. My local HAM repeater requires a key up every 15 minutes to avoid a disconnect.

How often does your 2nd problem occur? Please use the digital monitor from [here](https://github.com/DualTachyon/rt-4d-monitor/releases/tag/correct-timeout) and read the instructions [here](https://github.com/DualTachyon/rt-4d-monitor/?tab=readme-ov-file#how-to-use) on how to use it. The moment you see the problem appear press any key in the monitor to stop it from capturing, then copy and paste the last 20-30 lines over here. I recommend you set up the TG/etc you want to listen to first before starting the procedure detailed in the instructions. The reason for that is the PTT key shared the same pin as the data cable and if you PTT while the digital monitor is running, the data will most likely come out garbled until the cable is detached and the radio rebooted.

DualTachyon, 2025-05-12:
Someone gave me a capture that showed invalid DMR IDs being returned by the DMR firmware. I don't know if it's related to your point 2, but I made a change where "INVALID!" will be displayed if such an invalid value is detected. Give this binary a try.

[2025-05-12-102850.zip](https://github.com/user-attachments/files/20158325/2025-05-12-102850.zip)

DualTachyon, 2025-05-12:
This binary is has a better fix

[2025-05-12-151607.zip](https://github.com/user-attachments/files/20165857/2025-05-12-151607.zip)

Try it instead.

DualTachyon, 2025-06-06:
Guessing it all works now, closing.

closed 2025-06-06 by DualTachyon