DualTachyon/rt-4d-fw-beta #64: Beta 34.1: Wrong red button action.

Tov17, 2025-05-15:
When manual calling in call menu through green key, the long pressing of red cancel key works as action confirm instead of force exit.

DualTachyon, 2025-05-16:
Thanks. Let me know if the attached binary is better.

[2025-05-16-101006.zip](https://github.com/user-attachments/files/20241293/2025-05-16-101006.zip)

Tov17, 2025-05-17:
Yes, its much better, the confirming action was removed. But i feel the long press red button lost a feature. If i properly remember it had been exit.

DualTachyon, 2025-05-17:
> Yes, its much better, the confirming action was removed. But i feel the long press red button lost a feature. If i properly remember it had been exit into menu.

That was a side effect of the bug. Do you want me to restore long press for exir? It's easy to do.

Tov17, 2025-05-17:
Yes, it's need for instant exit into menu without ID deleting. 

Also it can be useful the id empting by red button long pressing then exit by short pressing again. 

I don't sure which case is better.

DualTachyon, 2025-05-18:
I think delete all digits is more useful, I really hate having to delete all 1 by 1.

DualTachyon, 2025-05-18:
Give this build a try.

[2025-05-18-171320.zip](https://github.com/user-attachments/files/20273698/2025-05-18-171320.zip)

Tov17, 2025-05-18:
I tested, it's wonderful )) i think it must be extendable into every input field.

closed 2025-05-18 by Tov17

DualTachyon, 2025-05-18:
I will try, but not today.