DualTachyon/rt-4d-fw-beta #67: Western Edition - error 433

PieeerB, 2025-06-03:
Install in VFO 433.000 and run the spectrum. The spectrum will be empty

[image]

DualTachyon, 2025-06-03:
This should fix it. It was introduced when I did a related change for another user.

[2025-06-03-185018.zip](https://github.com/user-attachments/files/20578518/2025-06-03-185018.zip)

PieeerB, 2025-06-03:
> This should fix it. It was introduced when I did a related change for another user.
> 
> [2025-06-03-185018.zip](https://github.com/user-attachments/files/20578518/2025-06-03-185018.zip)

Now everything is working, there is no error.

DualTachyon, 2025-06-03:
Thanks for the report. Closing.

closed 2025-06-03 by DualTachyon