DualTachyon/rt-4d-fw-beta #70: RSSI indicator error

MIF666, 2025-06-06:
[image]
[image]

1.RSSI - OK (mem 100)
2.RSSI - fault (mem101)

MIF666, 2025-06-07:
I didn't know where the error was, I sent the findings. But now I know :) When the power is H, it's fine and when it's L, the RSSI part is missing. When manually opening the monitor, it is fine even at L. When SQ is opened by a signal, the problem occurs.

DualTachyon, 2025-06-07:
OK, i will have a look.

MIF666, 2025-06-07:
H/L doesn't cause it. Sorry, it seemed that way. It's probably related to
some specific RSSI value. Somewhere around S6. I guess. I'll look into it
and let you know when it makes sense.

Sorry:(

MIF666, 2025-06-07:
[image]
[image]

The problem persisted and what's more, it's now a bad (positive) value. I will observe this and try to find deeper connections. Very thanks!

DualTachyon, 2025-06-07:
This suggests that the screen is getting cleared rather than RSSI being weird. The value should have stayed 3 digits at all times.

MIF666, 2025-06-07:
Don't waste your time with this. I'll try to be more specific about the
condition when this happens. I have 3 radios and they all do this,
including older FW versions.
Thx;)

DualTachyon, 2025-06-07:
You misunderstood what I said. The value on the screen is getting erased partially. that is a bug I need to fix. You don't need to figure out how to trigger, as now I know what to look for in the code.

MIF666, 2025-06-07:
Aaaa, OK, thx:)