DualTachyon/rt-4d-fw-beta #79: Group code entered by Green key short press is temporary and used only once

pawol, 2025-06-20:
When group code is entered by Green short press then the group is used only once . On next PTT that group is not used and valid. Each time we have to make additional shot Green key press before PTT.
I mean when have set some group eg. 4444 by means of zone selector submenu and you put a group eg. 1234 by Green key short press and start PTT then the group 1234 is used only once. Next press of PTT uses group 4444. Before each PTT we have to make Green key short press.
Extremally annoying behaviour. A group entered by Green key short press should be memorized and should overwrite a group selected by group selector submenu

DualTachyon, 2025-06-20:
That is the expected behaviour.. PTT by itself will always dial the default contact that you setup. You can change the default PTT contact at any time through the contacts list menu. I also added a "Contacts List" to the "key define" that you can use to quickly invoke it.

To make the green key contact override the default contact means that the default contact becomes largely useless. How do you go back to it?

pawol, 2025-06-20:
eg. on the selectable contact list should be something like "adhoc contact". when we make a contact by Green key short press (let's call it adhoc) then that "adhoc contact" should be selected automatically on contact selctor and overwritten by manually entered contact code. That is the solution.

DualTachyon, 2025-06-20:
OK... An ID of 0 is an invalid contact. If you make a contact with that ID=0, I can detect it and use the green key ID instead.

pawol, 2025-06-20:
I will be happy to test that featre because at the moment "adhoc" usability is terrible :)

crazy4volvo, 2025-06-20:
Great! I miss this feature a lot using my hotspot with no static group defined on TS1. I frequently forgot to set TG again with green key before PTT.

DualTachyon, 2025-06-20:
The display will probably be incorrect, but try this binary. Focus on the ID being sent (if you are using hotspot/repeater dashboard).
I need to check how to correct the display. Also, if you dial directly from main screen without using green key, might dial a 0. I'm still trying to figure it out.

To use, do as I described in my previous comment. Create a contact with ID of 0, then go to green key and enter your favourite combo.

[2025-06-20-221822.zip](https://github.com/user-attachments/files/20843185/2025-06-20-221822.zip)

crazy4volvo, 2025-06-20:
Shouldn't ad-hoc TG last until another channel is selected or another TG is set with green key?

DualTachyon, 2025-06-20:
Nevermind, I forgot to convert the contact number to the actual ID. there's a contact index into the contact database. i was looking at the index, not the actual ID.

DualTachyon, 2025-06-20:
Try this one

[2025-06-20-224008.zip](https://github.com/user-attachments/files/20843282/2025-06-20-224008.zip)

pawol, 2025-06-20:
I shall be back soon with a report.

pawol, 2025-06-20:
Does not work as planned. Still works in the old way

DualTachyon, 2025-06-20:
> Does not work as planned. Still works in the old way

Can you take a video?

pawol, 2025-06-20:
Can be a problem but nothing changed. Grenn short press do not memorize a code and there is nothing additional in the contact selector

DualTachyon, 2025-06-20:
And make sure you are using [2025-06-20-224008.zip](https://github.com/user-attachments/files/20843282/2025-06-20-224008.zip)
The previous binary I posted was definitely wrong.

pawol, 2025-06-20:
Still works like in the past

DualTachyon, 2025-06-20:
Take a video from boot, pressing PTT right away, then showing the default contact, then going to that contact and showing the ID.

DualTachyon, 2025-06-20:
That's not what i asked. I want to see the 4D boot into the main screen. At least 1 second of the boot screen, then press PTT on main screen. I suspect you're not using the right binary.

DualTachyon, 2025-06-20:
Also, your channel is set to contact ID 9990. That is not a contact ID of 0.

DualTachyon, 2025-06-20:
> I'm not sure I understand your question but what I am trying to do is:
> 
> 1. You must set default contact to one that has ID = 0
> 2. If you PTT directly without first going to green key, it will display an error.
> 3. Once you use green key, it will PTT as if a contact.
> 
> Currently the feature is not finished, but the dialing should work. The error may or may not display, and when you PTT, it might show the wrong contact, but what matters is what the dashboard displays... I'll fix the display+error in the meantime.

Follow the 3 steps above.

DualTachyon, 2025-06-20:
That's not a video.... I think you are not using the feature correctly anyway. You have to select a contact that has an ID of 0... then green key will be remembered temporarily.

pawol, 2025-06-20:
luke I said I cannot make a video now . In my case local 9 is the first editable contact in the cps

pawol, 2025-06-20:
nothing changes undel local 9 after green short press

DualTachyon, 2025-06-20:
If you don't change the default contact to a contact that has ID = 0, then nothing will change.

DualTachyon, 2025-06-20:
> aaa okay. I have to add new contact with the code 0. right?

yes

This new binary will clear the green ID between channels and A/B.

[2025-06-20-232002.zip](https://github.com/user-attachments/files/20843551/2025-06-20-232002.zip)

pawol, 2025-06-20:
Ok, now memorizes/adds new adhoc contact under "adhoc with id 0" but does not changes contact in the selector. I had to select adhoc manually from the list

DualTachyon, 2025-06-20:
You have to select the adhoc contact to be the default, otherwise it cannot work.
I cannot make the green key search for the adhoc contact and update that...
You have to change the contact when you want to use adhoc, and it will remember the last ID you enter on green key. You can change it 100 times, and it will remember the last one.

pawol, 2025-06-20:
Great, first test shows that works like expected. It was hard to me to understan what is ID=0. I thought that you mean first editable position in the list

DualTachyon, 2025-06-20:
The contact can be any and doesn't have to be the first one. The first editable one is index=1, because radtel hardcoded "ALL" to 0...
You can even have multiple adhocs, the firmware won't care.

pawol, 2025-06-20:
Super feature, now it is important, the people have to understand an idea and add one contact with ID=0 by default

pawol, 2025-06-20:
Maybe your cps should define that contact by default in the future

pawol, 2025-06-20:
Is it possible to distinguish privat and grou adhoc contact?

pawol, 2025-06-20:
something wrong when I added two contats with ID=0.now after green short press the key # does not change privat->group but shows "long press to remote wake"

DualTachyon, 2025-06-20:
That is normal.

[image]

DualTachyon, 2025-06-20:
Changing private has always been with the green key.

pawol, 2025-06-20:
sory, my fault, I should use green short press 2nd time to group, not #

pawol, 2025-06-20:
More tests tommorow, I do not think well now. bye

pawol, 2025-06-21:
Let's discuss again how this feature should work to be more user frendly...

IMHO. First usage of Geen short press including should create new position in the contact list  at first empty and available position with the hardcoded name "Adhoc contact". In case of missing free position a message "No space for Adhoc contact" should be displayed. All the time we are able of course to reuse Green short press and overwrite "Adhoc contact" when needed. Afterthat we can assign Green long press to the contact selector and easily switch onto newly created "Adhoc contact" and other ones. It will be much user freandly and easier to understand .

If someone will rename "Adhoc contact" then new one should be created at first empty and available position again. In that way we can quickly create new contacts.

DualTachyon, 2025-06-21:
I mentioned in the other thread but I don't want to change the main green key behaviour. But adding quick/easy "add contact" is no problem.

Adding flash lookup+saving code to the green screen is not a good idea in terms of code. It is better to redirect to the existing contact saving menu, something along those lines. The type+Id could be prepopulated, and maybe even the name, name=I'd, then the user can just press save, or change the name before saving.

pawol, 2025-06-21:
is it right behaviour when thecfeature does not work in case of mussing connection with repeater? In that case I see "connect to repeater failed" and entered adhoc contact is not memorized

DualTachyon, 2025-06-21:
Make sure you don't accidentally change channels or switch A<>B.

pawol, 2025-06-21:
By the way. cleaning ad-hoc contact when vfo is changed is not good option. It is better to keep adhoc all the time without cleaning

DualTachyon, 2025-06-21:
> By the way. cleaning ad-hoc contact when vfo is changed is not good option. It is better to keep adhoc all the time without cleaning

I let you debate this one with @crazy4volvo , because I think both opinions are contradictory (or I misunderstood)

pawol, 2025-06-21:
next pro for my proposal is fact we can always  delete/reedit code value when we make green short press if we want to change ad-hoc code

DualTachyon, 2025-06-21:
Star key is also available as an alternative, but to me using star breaks the pattern of how it is used in other screens. But I don't care too much either way.

pawol, 2025-06-21:
contact saving by pressing key F2 works. I have found athe bug. when radio works and adhoc feature was configured from the scratch and then I made firmware update, the feature stops to work. I had to select another contact from the list and then feature started to work again. I mean when I did fw update when adhoc contact was selected on the list then problem occured until I changed a contact from the list.

DualTachyon, 2025-06-21:
try to make a video, it is hard to follow the flow to trigger the bug.

Failte56, 2025-06-21:
From my point of view the Private call should work like on Motorola radios... Private call active between the two radios until the ESC button is pressed, which in the case of the 4D could be the orange button.....As the feature is now, the private call remains open for only one pass.

Failte56, 2025-06-21:
I write better, if I search in contacts I can have the private call without limits, but it would be better to have it from the green button....

Failte56, 2025-06-21:
If I use contacts, the private call remains active even if I turn it off and on again... I have to choose All Call to deactivate it

DualTachyon, 2025-06-21:
it would be best if each person here makes a simple table or list of the behaviours that you wish.

| Action  | Result |
| Press XYZ | Turn on washing machine |
| When ABC is DEF | Fold the laundry |

Then I can try to find common parts, and convert separate parts into different/separate features.

DualTachyon, 2025-06-21:
> | Action | Private call activation | | Press Green key | ID to be called for private call | | Press Orange Key | End private call an reactivate default contact |

What about the group call? Can it behave as was described by the new feature?

pawol, 2025-06-21:
@Failte56 remember that makes Green short press selects group. We should rather name is "adhoc calls" then "private calls"

Can it be applied in my way?

| Action | Private call activation |
| Short Press Green key | ID to be called for adhoc calls (private/group calls) |
| Press Red Key | Ends adhoc  calls an reactivate previously used contact from contact selector|

Failte56, 2025-06-21:
> [@Failte56](https://github.com/Failte56) remember that Green short press 2x selects group. We should rather name is "adhoc calls" then "private calls"
> 
> Can it be applied in my way?
> 
> | Action | Private call activation | | Short Press Green key | ID to be called for adhoc calls (private/group calls) | | Press Red Key | Ends adhoc calls and reactivates previously used contact from contact selector| | When radio power restarted | Reactivates previously used contact from contact selector |
> 
> So it looks that adhoc calls should be kept in RAM (not on the list)

For me it's ok,

pawol, 2025-06-21:
@DualTachyon is it possible to implement the feature in the way mensioned by me above?

DualTachyon, 2025-06-21:
is the new plan to remove the requirement for ID=0 contact?

DualTachyon, 2025-06-21:
@crazy4volvo can you clarify what you meant with your earlier comments?

DualTachyon, 2025-06-21:
Actually, star key can be enable and disable adhoc... I can display adhoc on the screen when it's enabled.

Give this a try.

[2025-06-21-195256.zip](https://github.com/user-attachments/files/20848292/2025-06-21-195256.zip)

DualTachyon, 2025-06-21:
This one doesn't corrupt the screen as much :)

[2025-06-21-195544.zip](https://github.com/user-attachments/files/20848303/2025-06-21-195544.zip)

DualTachyon, 2025-06-21:
With the last binary, ID=0 contact is no longer needed.

pawol, 2025-06-21:
Now rt-4d works perfectly. * key after Green short press is very comfortable. I confirm it works okay

[12 more replies]

closed 2025-06-22 by DualTachyon