DualTachyon/rt-4d-fw-beta #88: signal meter shows a bit ubelievable values of backround noise

pawol, 2025-06-24:
bug - signal meter shows unbelievable values of background noise.
I saw it with each whils Airband AM reception.
Very often a background noise level I saw was -156 (dBm?)  when modulation dissapeared. I predict in that siltation a level close to -125 dBm rather then -156 dBm.

That value is raher unbelievable

pawol, 2025-06-24:
yeah, That is it. I have checked signal readings by means of tinySA. Reading in FM mode are right but in AM mode are too low. Look on attached pictures made in AM/FM and different levels

[image]
[image]
[image]
[image]
[image]
[image]

pawol, 2025-06-25:
ok. results soon.

pawol, 2025-06-25:
worse than before. Now in AM shows:
-48 instead of -116
-28 instead of -96
-27 instead of -76

etc.

DualTachyon, 2025-06-25:
OK, then I don't know what it does...

pawol, 2025-06-25:
in the same time on FM
-117 s5  when -116
-96 s7 when -96
-75 s9 when - 76

pawol, 2025-06-25:
should be the same in my opinion. Ssb can be different because without modulation there is no signal

DualTachyon, 2025-06-25:
It cannot be the same, because AM has dynamic AGC adjustments based on RSSI measurement to improve the audio quality.

pawol, 2025-06-25:
aaaa. AGC. hmmm. we need to be able to disale AGC for the test. can you add AGC off test option?

DualTachyon, 2025-06-25:
AGC disabled and RSSI is not adjusted.

[2025-06-25-232942.zip](https://github.com/user-attachments/files/20912948/2025-06-25-232942.zip)

DualTachyon, 2025-06-25:
As my previous comment said: "it is largely a blackbox to me. I don't understand why it does what it does."

pawol, 2025-06-25:
ok. at the moment is it better to revert changes until you will understand the code