DualTachyon/rt-4d-fw-beta #90: 2 Issues, RX cuts out.. & DMR Scanning

ICantMakeThings, 2025-06-27:
There are 2 issues i found from using the firmware 39 for a few days, when i have a radio close enough in NFM mode, it cuts out when i speak too loudly, Not the case when in WFM or DMR mode. (Using newest DMR version and FW version)

2nd issue is not exactly a issue, but scanning in channel mode is very slow over DMR channels, In settings it only goes down to 400ms, I'm only complaining since i have another radio with OpenGD77 and it goes 10x faster (or 5, idk) so its doable, Is it possible to add a faster speed option?

Except from that, firmware is top notch!

I made a video showing how to flash your firmware 
https://www.youtube.com/watch?v=W1mcqlKWJXw
If you want to you can add it in to your readme, if not thats fine, just mentioning it is all.

DualTachyon, 2025-06-27:
It is not possible to go lower than 400ms for DMR scanning. 400ms is already sketchy at times. The DMR firmware requires a lot of time to detect a valid voice stream, which is only slightly more than DMR super frame:

```
Voice transmissions use a superframe that is 6 bursts (360 ms) long with bursts labelled "A" to "F"
```

I'm not sure what you mean with the RX cut out problem?

ICantMakeThings, 2025-06-27:
Interesting, wonder how OpenGD77 does it, anyway.
What i mean is if i have another radio transmitting, both on narrow FM, if i get the radio within 1m, when i press ptt it receives, when some noise is made or i speak it clips out, but like 2m away it works just fine. not a huge issue but no other radios i own do that.

Also got a suggestion, is it possible to add more precise power control, instead of Low and High?

DualTachyon, 2025-06-28:
OpenGD77 radios have a register-programmable DMR chip that gives you full control. The RT-4D uses a fake-DMR chip that runs its own firmware that controls the BK4819 directly. My firmware only has high level access to it.

What you describe about RX sounds like an RF problem, probably of being too close. Either way, it is outside of what firmware can influence.

TX power is not linear and [Radtel recommends](https://github.com/DualTachyon/rt-4d-fw-beta/discussions/60) not touching the gain values.

ICantMakeThings, 2025-06-28:
okay, thanks anyway for your replies, best wishes

closed 2025-06-28 by ICantMakeThings